analogdevicesinc/ad5940-examples #2: IAR download failed.

cybertale, 2019-08-17:
Following the steps from [How to setup and use IAR Embedded Workbench](https://wiki.analog.com/resources/eval/user-guides/eval-ad5940/tools/iar_setup_guide), I'm failing to download the example programs into the ADUCM3029 board. It always shows error message 'Failed to load flash loader'.
I got one guess but not sure about it. I have two ADUCM3029 boards and all cannot be downloaded by IAR after I update their firmware with SensorPal.
For now I download example programs with sending the object file to the folder that DAP-Link simulated.
This isn't an important issue for me, just want to give you a report of the problem.

neoanalog, 2019-09-19:
This issue has no final solution yet. There are some solutions here that can be tried.
1. Erase flash content
https://wiki.analog.com/resources/eval/user-guides/eval-ad5940/help_and_support
2. Use on board debugger's drag-and-drop programming function to flash a new hex file to ADuCM3029. https://ez.analog.com/data_converters/precision_adcs/f/q-a/113742/ad5940-example-failed-to-load-flash-loader/341992#341992

flowersnlyrics, 2019-09-23:
The IAR build doesn't work.

flowersnlyrics, 2019-11-04:
@neoanalog when will this issue have a final solution?

neoanalog, 2019-11-07:
Hi @mlambe, do we have update on this issue? If IAR is still failing, try to use Keil, the evaluation license should be enough for most examples. The version number of IAR passed our test is 8.32.1.

mlambe, 2019-11-08:
Hi @flowersnlyrics what build errors do you get on IAR? Have you followed the setup instructions on the user guide [here](https://wiki.analog.com/resources/eval/user-guides/eval-ad5940/tools/iar_setup_guide)