analogdevicesinc/ad5940-examples #3: Embedded Recursion

flowersnlyrics, 2019-09-27:
Why is there recursion in AD5940_BIA project? And to set up the clock in the ad5940lib? I feel as though the same state machine could achieved without it and recursion is usually discouraged on embedded devices.

neoanalog, 2019-10-02:
The function AD5940_ClksCalculate is recursive, it's only used for calculating number of clocks used and the maximum recursion depth is 3 i think.

flowersnlyrics, 2019-10-02:
So is AppBIACtrl in BodyImpedance.c of the AD5940_BIA example

flowersnlyrics, 2019-10-02:
Have you maybe thought of putting the state machine in an infinite loop and having a case in your switch statement that breaks out of that loop? That kind of protects the stack from a recursion gone rogue situation.

neoanalog, 2019-10-12:
The recursive function will exit in a certain loop. For example, AppBIACtrl(BIACTRL_SHUTDOWN) will call this function again with parameter AppBIACtrl(BIACTRL_STOPNOW). It will exit immediately, thus no risk of run out of stack.

neoanalog, 2019-10-21:
Hi,
We have examples on board NUCLEO-F411RE. The file you need to change is NUCLEOxxxPort.c if different board is used. You can post [here](https://ez.analog.com/data_converters/precision_adcs/f/q-a) if need any help. I will close this issue and you are welcome to start a new one if any issue found. Thanks.

closed 2019-10-21 by neoanalog