analogdevicesinc/ad5940-examples #6: TODO in APP_BIA

flowersnlyrics, 2019-11-04:
https://github.com/analogdevicesinc/ad5940-examples/blob/dfb9914a12f342d6eb110f28ee0bab800dc11bd8/examples/AD5940_BIA/BodyImpedance.c#L115

XuNeo, 2019-11-07:
Hi,
The comment should be "it's not sure this will stop the WUPT, better to use BIACTRL_STOPSYNC". This example uses sequencer and let AFE goes to hibernate mode periodically, thus there is chance that AFE goes to hibernate mode just before line 115 is executed, no matter if you wake up AFE in line 111 or not. However, the chance is small, so we leave the code there. Will update comments soon...
Thanks.
Neo

neoanalog, 2019-11-07:
Hi,
The comment should be "it's not sure this will stop the WUPT, better to use BIACTRL_STOPSYNC". This example uses sequencer and let AFE goes to hibernate mode periodically, thus there is chance that AFE goes to hibernate mode just before line 115 is executed, no matter if you wake up AFE in line 111 or not. However, the chance is small, so we leave the code there. Will update comments soon...
Thanks.
Neo