analogdevicesinc/ad5940-examples #9: AD5940_Impedance example not working

paulrox, 2020-02-07:
Hi,

I have some problems running the AD5940_Impedance example.

I'm using the AD5940 from a host MCU through SPI interface for which I already implemented and tested the necessary functions to port the ad5940lib. I have also tested the AD5940_ECSns_EIS example using a conductivity probe and it seems to work very well.

From what I've seen, the measure of the impedance is almost identical to the one used in the AD5940_ECSns_EIS except for the fact that in the AD5940_Impedance example the R_LOAD resistor on SE0 is not used for the measure. Looking at the comments on the code, I thought that using the dummy sensors on the EVAL-AD5940ELCZ board I should have read a value near 5KOhm while in reality, I read a value which is over 10KOhm. In addition, I saw that the switch matrix is configured to use the SE0 as input for the TIA but this would require the use of the calculations used in the AD5940_ECSns_EIS example.

There is something wrong in the code or am I missing some steps?

neoanalog, 2020-02-10:
Hi,
I'm not sure how you connect the conductivity probe, but the AD5940_Impedance example should work on EVAL-AD5940ELCZ. It's high possible you have wrong configuration on the jumpers. Take a look at [5940's wiki page](https://wiki.analog.com/resources/eval/user-guides/eval-ad5940/software_examples/ad5940_impedance#setting_up_the_hardware). Follow the instructions except step5 where you should use position B on JP6 to select onboard 5kOhm. Schematic is [here ](https://wiki.analog.com/_media/resources/eval/user-guides/eval-ad5940/hardware/02_050193a_top.pdf) for your reference.

The main difference between ECSns_EIS and Impedance is the DUT impedance range.
AD5940 can not measure impedance in Ohm range directly.
So, ECSns_EIS measure impedance of _Zdut+Rload_ then _Rload_. Then _Zdut+Rload_ - _Rload_ equals to Zdut.

paulrox, 2020-02-10:
Hi,

Thanks for your response.

After checking the example code I found a mistake in the porting of the example to my platform.
Anyway, in the wiki page that you linked, the screenshot of the serial terminal shows a magnitude of around 680 Ohm, while the EVAL-AD5940ELCZ has a dummy sensor of 6.8 KOhm (which is also the value which I read by running the example).

I didn't get the point of this part of your response:

> The main difference between ECSns_EIS and Impedance is the DUT impedance range.
> AD5940 can not measure impedance in Ohm range directly.
> So, ECSns_EIS measure impedance of Zdut+Rload then Rload. Then Zdut+Rload - Rload equals to Zdut.

From what I understood, both examples use a ratiometric approach to compute the impedance of the sensor. The only difference is that the ECSns_EIS example uses the R_LOAD (which is attached to the SE0 pin) in the computations while the impedance example does not. I extracted this information from the AN-1563 application note. The only thing which I don't understand here is why the R_LOAD value is not used in the Impedance example even if the SE0 is being used as sense electrode.

neoanalog, 2020-02-10:
Hi,
> both examples use a ratiometric approach to compute the impedance of the sensor.

Yes, but only when impedance is roughly higher than 100Ohm, otherwise, you need one additional step which is exactly described in AN-1563. Rload makes sure AD5940 measures impedance >100Ohm.

>The only thing which I don't understand here is why the R_LOAD value is not used in the Impedance example even if the SE0 is being used as sense electrode.

SE0 is not  used as the sense electrode in ECSns_EIS but the node _SWN_SE0LOAD_.  SWN_SE0 is used in Impedance example, so the kelvin connection makes sure Rload is excluded.

paulrox, 2020-02-10:
Hi,

Thanks for the quick reply.

> Yes, but only when impedance is roughly higher than 100Ohm, otherwise, you need one additional step which is exactly described in AN-1563. Rload makes sure AD5940 measures impedance >100Ohm.

Ok, so if I have to measure a low impedance sensor (e.g. a Pt100 RTD) I should use the R_LOAD to ensure the quality of the measure?

> SE0 is not used as the sense electrode in ECSns_EIS but the node SWN_SE0LOAD. SWN_SE0 is used in Impedance example, so the kelvin connection makes sure Rload is excluded.

Can you please explain better this part? When SWN_SE0  is used the R_LOAD is excluded from the N path to the excitation amplifier but is still present on the path to the TIA through the switch T5.

neoanalog, 2020-02-11:
Hi,
An easy method to determine if DUT impedance is to check if the excitation current excess HSTIA limitation(3mA).
Considering you are measuring PT100, you would better to use lower excit-volt to minimize self heating. So, you could measure it directly without R_LOAD.
 
The RLoad is in path of 'force' signal, so it won't affect your measurement. Check it [here ](https://en.wikipedia.org/wiki/Four-terminal_sensing)about force/sense.

I will close this issue now. You are welcome to post any further questions in [EngineerZone](https://ez.analog.com/).

closed 2020-02-11 by neoanalog