analogdevicesinc/ad5940-examples #20: Question about BATimpedance

tomm1gun, 2022-07-21:
Hi, within the BATimpedance code(AD5940_BATImpedance) there is a function(AD5940BATStructInit) that allows the user to modify some parameters.
One of them is pBATCfg.DCVolt which is set to 1200.0f, so what does this voltage refer to?
At the input of the battery there is a current not a voltage.

lp_loop.LpDacCfg.DacData12Bit = (uint32_t)((AppBATCfg.DCVolt-200)/2200.0f*4095);

it would be helpful to comment on this line.
Thanks.

XuNeo, 2022-07-21:
This DC voltage sets DC current drawn from battery. 
To measure impedance, AC current is drawn through battery and it excites the internal battery impedance. 
There is no charge capability provided, thus this DC current is needed. Similar to the AC voltage on top of DC offset when dealing with single power supply circuit.

tomm1gun, 2022-07-22:
i get it, I looked on the datasheet the part about the power supply but I didn't understand, where can I find those 1.2 volts on the circuit? what is the relationship between voltage and current? I am taking some measurements but they are very noisy, i'm trying to understand where is the problem.

XuNeo, 2022-07-26:
[image]

Check CN0510, the voltage should show up on pin AIN4. Note the switch matrix N is connected to AIN4, so the DC offset on pin CE0 is also set.

closed 2022-07-28 by tomm1gun

XuNeo, 2023-09-04:
@znoorsa  It's better to ask for help on their website `Engineer zone` where it's more likely you get official response from.
https://ez.analog.com/

znoorsa, 2023-09-12:
I will leave my Question in that forum , Thank you for your Help 🙏